RoboJackets/robocup-pcb #15: Create pcb for 2015 base station

justbuchanan, 2016-06-13:
This won't be anything fancy, just something simple so we don't have to use in place of a breadboard.

cc @jjones646

jjones646, 2016-06-13:
probably will want to support 2 radios on separate spi busses.

justbuchanan, 2016-06-13:
Good call.  We talked about doing that before, but I had completely forgotten about it.

justbuchanan, 2016-06-14:
@jjones646 any idea where I can find an eagle library for this part?

http://www.digikey.com/product-search/en/connectors-interconnects/rectangular-board-to-board-connectors-headers-male-pins/1442672?k=609%203175%20nd

justbuchanan, 2016-06-14:
Also, what's the best way to show connections to the mbed in the schematic?  The board will have the female headers for the mbed to plug into, but it would be nice in the schematic to show the pinout of the mbed.  I guess I could just add the headers to the schematic and label the relevant pins, but I'm not sure if there's a better way to use the mbed.

jjones646, 2016-06-14:
make it :smile:. It's a super easy part, just make the pads shorter than what the datasheet recommends...

There should be an mbed library part in the repo somehwere

justbuchanan, 2016-06-15:
Sounds good, I should be able to handle that.

Yeah I see the mbed library in this repo.  My question was more of should I put the headers in the schematic or the mbed itself?  Or both?  The pcb will just have two rows of headers mounted on it.

jjones646, 2016-06-16:
Just the mbed in the schematic should work out fine.

PR #18 (2016-07-10): Cleanup with base station files
Merged things together manually with the cleaned out pr I have on here.

fixed #15 
closes #16
closes #17
closed 2016-07-10 by jjones646