RoboJackets/robocup-pcb #21: Update Control Board to Spartan 7

guyfleeman, 2018-05-25:
We should work toward a control board that supports the Spartan 7 for a few reasons.

1. ISE is approaching EoL/EoS
2. The newest version of Verilog supported is 2001 (17 years ago)
3. No SV Support
4. Outdated Xilinx IP

The candidate replacement chip seems to be the Spartan 7 [XC7S25](https://www.digikey.com/product-detail/en/xilinx-inc/XC7S25-1FTGB196C/122-2122-ND/8040787) which is several times more powerful than the current [Spartan 3E - XC3S250E](https://www.xilinx.com/publications/prod_mktg/spartan-3-product-table.pdf) that we use.

Desired Added Feature List
- [ ] Spartan 7
- [ ] Current Sensing on Motor Phase Lines

guyfleeman, 2018-05-25:
So the Spartan 7 is 196-LFBGA. What would be a cost estimate per board if we did this at M-Train quality?

petersonev, 2018-05-25:
Looks like the pitch is 1mm instead of 0.8 so we might be able to get away with a cheaper classification. I'll check eurocircuits to find the exact cost. But upgrading the classification on eurocircuits is actually not that much, especially with large quantities.

guyfleeman, 2018-05-25:
I'm comfortable trying to port the current control scheme to the 7 but probably not adding the current sense.

guyfleeman, 2018-05-25:
@petersonev what are you interested in working on? Or are you sticking with M-Chair software rn?

petersonev, 2018-05-25:
I'll keep working on the new MCU board stuff but I'll at least make a PCB with the spartan 7 so that can get ordered first thing in the year. I'll definitely try to help on verilog stuff too (as much as I can) because I want to get better at that

guyfleeman, 2018-05-25:
So what do you want me to work on first/rn? SV sim or Spartan 7 port?

petersonev, 2018-05-25:
SV sim is probably better to get started on first, especially because you have a lot more experience with system verilog

guyfleeman, 2018-05-25:
Okay. I almost want to try routing that board tho when the schematic is done. Defs keep me in the loop

petersonev, 2018-05-25:
On eurocircuits we should actually be able to still use class 6 and just upgrade the drill size to 0.15mm so that's only like $10 extra total which is basically nothing. I may still upgrade to class 7 so the routing can be nicer because that's only like $30 extra total. I'll have exact prices later but the change in PCB cost should be negligible

guyfleeman, 2018-05-25:
Good

closed 2022-05-14 by Arvind-Srinivasan